jkjung-avt/tensorrt_demos #537: Segmentation fault when closing window or using ctrl+C

Duarte-Nunes, 2022-03-24:
Howdy, this is just a comment not exactly a breaking issue.

When i close the opencv window with the mouse or interrupt the code with ctrl+C i get this message:

[03/24/2022-11:42:12] [TRT] [E] 1: [defaultAllocator.cpp::deallocate::35] Error Code 1: Cuda Runtime (invalid argument)
Segmentation fault (core dumped)

But for example when i use the ESC key or if im using a video file and the video ends, the code closes without an error.

Is this related to maybe OpenCV?

jkjung-avt, 2022-03-26:
This problem seems to be an issue of "not handling GPU resource cleanup properly".

Please specify which demo you are testing with.

Duarte-Nunes, 2022-03-28:
Hello, im using trt_yolo.py in Demo 5.

jkjung-avt, 2022-04-03:
I briefly looked at it and think it's not easy to fix in current code base.

The user could hit Ctrl-C at any point of time, including when [context.execute_async_v2()](https://github.com/jkjung-avt/tensorrt_demos/blob/c6afc40082da72397b2b0ad91eb6f32acf3b731d/utils/yolo_with_plugins.py#L262) is being executed.  I'm not sure whether it's possible to terminate the tasks and clean up resources properly for all cases...

I'll leave the code as it is now.

jkjung-avt, 2022-04-05:
Your note for future work is well taken.  Thanks for letting my know this repo is helpful to you.

I'll close this issue, since I will not be tracking it.

closed 2022-04-05 by jkjung-avt