jkjung-avt/tensorrt_demos #539: onnx 1.4.1 dependency

tnaduc, 2022-03-28:
Hi Jk,
Thank you for the tutorials. They are extremely helpful.

Could you please help to look into the onnx==1.4.1 dependency for the yolov4/yolov3 demo and shine some lights on how can we overcame this dependency to use later onnx version to potentially open up to more models.

I understand it is mentioned here:
https://forums.developer.nvidia.com/t/tensorrt-backend-for-onnx-on-jetson-nano/74980/6

jkjung-avt, 2022-03-31:
This onnx==1.4.1 requirement is due to [the use of ONNX "Upsample" op](https://github.com/jkjung-avt/tensorrt_demos/blob/67c4eb0f5e8a46656eb65073e5685dc7654498d8/yolo/yolo_to_onnx.py#L908-L914) to implement "upsample" layer in DarkNet models, e.g. [a example "upsample"](https://github.com/AlexeyAB/darknet/blob/master/cfg/yolov4.cfg#L825-L826) layer in the yolov4 model.

The ONNX "Upsample" op is only available in python onnx==1.4.1 (ONNX opset 9).  It has been [deprecated since ONNX opset 10](https://github.com/onnx/onnx/issues/2501).  That's why my current code in this repo only works with onnx==1.4.1.

You could reference NVIDIA's latest [TensorRT/samples/python/yolov3_onnx](https://github.com/NVIDIA/TensorRT/tree/main/samples/python/yolov3_onnx) code for how to solve this.  It has been updated to [onnx==1.9.0](https://github.com/NVIDIA/TensorRT/blob/46253b644142a1d9632ba463422abfc5dcefc371/samples/python/yolov3_onnx/requirements.txt#L4).  And it uses [ONNX "Resize" op](https://github.com/NVIDIA/TensorRT/blob/46253b644142a1d9632ba463422abfc5dcefc371/samples/python/yolov3_onnx/yolov3_to_onnx.py#L693-L701) to implement DarkNet's "upsample" layer instead.

I might look into this and update the code when I have time...

jkjung-avt, 2022-05-06:
@tnaduc  I've updated the "yolo_to_onnx.py" code.  It now works for onnx==1.9.0 (but won't work for onnx==1.4.1 anymore).  I've only tested it with TensorRT 8.0 so far, but I think it should work for TensorRT 7 as well.

Please give it a try.

closed 2022-05-21 by jkjung-avt

tnaduc, 2022-06-20:
Thank you @jkjung-avt you are amazing.