jkjung-avt/tensorrt_demos #542: Parameters count for the optimized network

dl-infinity, 2022-04-07:
Thank you for your excellent work! After optimizing the network to tensorrt, how can we obtain the number of parameters in the network?

jkjung-avt, 2022-04-08:
I don't know whether there's an API call in TensorRT library for checking that.  Maybe you could raise the question on [NVIDIA Developer Forum](https://forums.developer.nvidia.com/c/ai-data-science/deep-learning/tensorrt/92) instead.

dl-infinity, 2022-04-09:
@jkjung-avt thank you! I have raised the question on the NVIDIA Developer Forum. From your experience, do you think it is possible? I have not seen existing questions related to it.

jkjung-avt, 2022-04-10:
> From your experience, do you think it is possible?

Certainly.  For example, when you call TensorRT library to load an engine to do inference, it needs to deserialize the model and load all parameters to GPU memory.  I think it a matter of whether NVIDIA has made such an API public.

dl-infinity, 2022-04-10:
Thank you! Would it be possible for you to look into such an API? I looked for the API over here [](https://docs.nvidia.com/deeplearning/tensorrt/api/index.html )but do not quite understand if there is some function available or not. I also tried to check all methods for the **trt_yolo** object on which we call **_detect_** method, but none of them seems to be able to count parameters.

If such an API is not available, could you suggest some ways to manually count the number of parameters in Python?

jkjung-avt, 2022-04-11:
If you don't find anything in [TensorRT API](https://docs.nvidia.com/deeplearning/tensorrt/api/index.html), just try to get an answer from NVIDIA Developer Forum.  I don't have any better answer...

closed 2022-04-25 by jkjung-avt