jkjung-avt/tensorrt_demos #547: Error in onnx to tensorrt. cuda failure in file 'yolo_layer.cu' line 51:35 Aborted (core dumped) Request assistance from the big guys

wangyu96-art, 2022-05-03:
Error in onnx to tensorrt.
cuda failure in file 'yolo_layer.cu' line 51:35
Aborted (core dumped)
Request assistance from the big guys

wangyu96-art, 2022-05-03:
Using a custom yolov3 model after cropping

wangyu96-art, 2022-05-03:
CHECK(cudaMalloc(&mAnchors, MAX_ANCHORS * 2 * sizeof(float)));

jkjung-avt, 2022-05-03:
Please provide your cfg file.  I will take a look.

wangyu96-art, 2022-05-06:
[converted.zip](https://github.com/jkjung-avt/tensorrt_demos/files/8638228/converted.zip)

jkjung-avt, 2022-05-06:
Your cfg file looks OK to me.

Failure to execute this line of code means the program cannot allocate memory on the GPU.

```C
CHECK(cudaMalloc(&mAnchors, MAX_ANCHORS * 2 * sizeof(float)));
```

Which GPU are you using?  Do you have multiple GPUs on the system?  If so, try to set, say, `CUDA_VISIBLE_DEVICES=0` (environment variable) to see if it helps.

wangyu96-art, 2022-05-08:
I'm using a Jetson nano development board, I'll try your help first, thanks.

jkjung-avt, 2022-05-09:
There should be only 1 GPU on your Jetson DevKit...

I wonder whether there's a hardware problem on your Jetson, or maybe most of the system memories have been occupied by other programs.

closed 2022-05-21 by jkjung-avt