jkjung-avt/tensorrt_demos #549: Unable to install onnx 1.4.1 in the virtual environment on Jetson AGX Xavier

sachinkmohan, 2022-05-04:
System Details
- - - 
- Jetson AGX Xavier
- Jetpack - 4.6
- Ubuntu - 18.04
- Virtual environment - venv
- pip version  - 21.3.1

I am unable to install the onnx version 1.4.1 as mentioned in this repository in the virtual environment. However, when I do `pip install onnx` , it installs the new version of onnx without any problem. Could you please check.

```python
(jkjung_yolo_1) user@user-desktop:~$ pip install onnx==1.4.1
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Looking in indexes: https://pypi.org/simple, https://pypi.ngc.nvidia.com
Collecting onnx==1.4.1
  Downloading onnx-1.4.1.tar.gz (2.9 MB)
     |████████████████████████████████| 2.9 MB 4.9 MB/s            
  Preparing metadata (setup.py) ... done
Requirement already satisfied: protobuf in /usr/local/lib/python3.6/dist-packages (from onnx==1.4.1) (3.19.4)
Requirement already satisfied: numpy in /usr/local/lib/python3.6/dist-packages (from onnx==1.4.1) (1.19.4)
Requirement already satisfied: six in ./virtual_env/jkjung_yolo_1/lib/python3.6/site-packages (from onnx==1.4.1) (1.16.0)
Requirement already satisfied: typing>=3.6.4 in ./virtual_env/jkjung_yolo_1/lib/python3.6/site-packages (from onnx==1.4.1) (3.7.4.3)
Requirement already satisfied: typing-extensions>=3.6.2.1 in ./virtual_env/jkjung_yolo_1/lib/python3.6/site-packages (from onnx==1.4.1) (4.1.1)
Using legacy 'setup.py install' for onnx, since package 'wheel' is not installed.
Installing collected packages: onnx
    Running setup.py install for onnx ... error
    ERROR: Command errored out with exit status 1:
     command: /home/user/virtual_env/jkjung_yolo_1/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-_g5i54n9/onnx_0d66111426534ee5b5c48c7a0d7063c8/setup.py'"'"'; __file__='"'"'/tmp/pip-install-_g5i54n9/onnx_0d66111426534ee5b5c48c7a0d7063c8/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-j4v5fy6h/install-record.txt --single-version-externally-managed --compile --install-headers /home/user/virtual_env/jkjung_yolo_1/include/site/python3.6/onnx
         cwd: /tmp/pip-install-_g5i54n9/onnx_0d66111426534ee5b5c48c7a0d7063c8/
    Complete output (65 lines):
    fatal: not a git repository (or any of the parent directories): .git
    WARNING: The wheel package is not available.
    running install
    running build
    running build_py
    running create_version
    running cmake_build
    -- Build type not set - defaulting to Release
    -- The C compiler identification is GNU 7.5.0
    -- The CXX compiler identification is GNU 7.5.0
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Check for working C compiler: /usr/bin/cc - skipped
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Check for working CXX compiler: /usr/bin/c++ - skipped
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    CMake Error at CMakeLists.txt:217 (message):
      Protobuf compiler not found
    Call Stack (most recent call first):
      CMakeLists.txt:248 (relative_protobuf_generate_cpp)
    
    
    -- Configuring incomplete, errors occurred!
    See also "/tmp/pip-install-_g5i54n9/onnx_0d66111426534ee5b5c48c7a0d7063c8/.setuptools-cmake-build/CMakeFiles/CMakeOutput.log".
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-_g5i54n9/onnx_0d66111426534ee5b5c48c7a0d7063c8/setup.py", line 328, in <module>
        'backend-test-tools = onnx.backend.test.cmd_tools:main',
      File "/usr/local/lib/python3.6/dist-packages/setuptools/__init__.py", line 163, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.6/distutils/command/install.py", line 589, in run
        self.run_command('build')
      File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/usr/lib/python3.6/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-_g5i54n9/onnx_0d66111426534ee5b5c48c7a0d7063c8/setup.py", line 203, in run
        self.run_command('cmake_build')
      File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-_g5i54n9/onnx_0d66111426534ee5b5c48c7a0d7063c8/setup.py", line 190, in run
        subprocess.check_call(cmake_args)
      File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/usr/local/bin/cmake', '-DPYTHON_INCLUDE_DIR=/usr/include/python3.6m', '-DPYTHON_EXECUTABLE=/home/user/virtual_env/jkjung_yolo_1/bin/python3', '-DBUILD_ONNX_PYTHON=ON', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DONNX_NAMESPACE=onnx', '-DPY_EXT_SUFFIX=.cpython-36m-aarch64-linux-gnu.so', '/tmp/pip-install-_g5i54n9/onnx_0d66111426534ee5b5c48c7a0d7063c8']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/user/virtual_env/jkjung_yolo_1/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-_g5i54n9/onnx_0d66111426534ee5b5c48c7a0d7063c8/setup.py'"'"'; __file__='"'"'/tmp/pip-install-_g5i54n9/onnx_0d66111426534ee5b5c48c7a0d7063c8/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-j4v5fy6h/install-record.txt --single-version-externally-managed --compile --install-headers /home/user/virtual_env/jkjung_yolo_1/include/site/python3.6/onnx Check the logs for full command output.

```

sachinkmohan, 2022-05-05:
Apologies, looks like my pip is broken somehow. I checked on my other jetson AGX board and it worked.

closed 2022-05-05 by sachinkmohan

jkjung-avt, 2022-05-06:
https://github.com/jkjung-avt/tensorrt_demos/issues/539#issuecomment-1119218462