jkjung-avt/tensorrt_demos #552: Converting old plugin to new plugin

keremyldrr, 2022-05-11:
Hello, 
I am trying to use the yolo layer implementation in this repository with the network implementation at https://github.com/tjuskyzhang/Scaled-YOLOv4-TensorRT/tree/master/yolov4-p5-tensorrt.

I have seen that their used plugin is one of the references the one in this repository was built upon. The difference is, the one here processes the 3 yolo layers one by one, while in their version they process all at the same time. When comparing the outputs, it looks like one of the boxes is incorrect(see pictures attached). Which led me into thinking that the cause is potentially the way the anchors are handled. Could you give any pointers to how to address this issue? 

These are the anchors of the yolov4p5 implementation, 8 for each layer:
 13,17,  31,25,  24,51,  61,45,
 48,102,  119,96, 97,189, 217,184,
 171,384, 324,451, 616,618, 800,800
 
While in the original yolov4 version, the anchors were 6 for each layer:
 12,16, 19,36, 40,28 
 36,75, 76,55, 72,146 
 142,110, 192,243, 459,401
 
Also, for clarification, first image the result of the implementation with the older plugin: 
[image]

As reference, the original pytorch implementation's(https://github.com/WongKinYiu/ScaledYOLOv4) results are like below:
[image]

Second image is the result of the implementation with the yolo layer in this repository(using their anchors)
[image]

All versions use the same pretrained weights.

jkjung-avt, 2022-05-21:
Could you provide the cfg/weights files for reproducing this issue?  I'll check it when I have time.

jkjung-avt, 2022-06-09:
@keremyldrr  Could you provide the cfg/weights files for reproducing this issue?

keremyldrr, 2022-06-14:
@jkjung-avt  Sorry for the late reply, i did not see the notification. I have fixed the issue, the reason was when using  `newCoords = 1` argument, the plugin was assuming network logits from previous layer has been passed through sigmoid activation, which was not the case in my application. When i included the sigmoid inside the cuda kernel that does the computation, everything behaved as expected.