jkjung-avt/tensorrt_demos #553: Assertion error while converting MobileNet V2 to TRT

wadhwasahil, 2022-05-17:
I am trying to follow [GitHub - jkjung-avt/tensorrt_demos: TensorRT MODNet, YOLOv4, YOLOv3, SSD, MTCNN, and GoogLeNet](https://github.com/jkjung-avt/tensorrt_demos) to convert custom trained SSD Mobilenet V2 model to TensorRT format. I get this error while running the conversion.

#assertion/opt/tensorrt/TensorRT/plugin/nmsPlugin/nmsPlugin.cpp,246

I have tried changing the inputOrder on this line https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/ssd/build_engine.py#L58 to [0, 2, 1] as the pbtxt file generated contains the following format

id: "NMS"
    inputs: "Squeeze"
    inputs: "concat_priorbox"
    inputs: "concat_box_conf"
Environment
TensorRT Version: 7.0.0
Tensorflow. Version: 1.15

jkjung-avt, 2022-05-17:
Refer to: https://github.com/jkjung-avt/tensorrt_demos/issues/149#issuecomment-654169120

Make sure you add 1 (for 'background') to `num_classes` for your custom model.

For example, for the Egohands model,  `num_classes` = 1 ('hand') + 1 ('background') = `2`.

https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/ssd/build_engine.py#L67

wadhwasahil, 2022-05-17:
@jkjung-avt I have already added 1 to my num_classes. Is there something else?

wadhwasahil, 2022-05-17:
```
"""build_engine.py

This script converts a SSD model (pb) to UFF and subsequently builds
the TensorRT engine.

Input : ssd_mobilenet_v[1|2]_[coco|egohands].pb
Output: TRT_ssd_mobilenet_v[1|2]_[coco|egohands].bin
"""

import os
import ctypes
import argparse

import numpy as np
import uff
import tensorrt as trt
import graphsurgeon as gs

DIR_NAME = os.path.dirname(__file__)
LIB_FILE = os.path.abspath(os.path.join(DIR_NAME, 'libflattenconcat.so'))
MODEL_SPECS = {
    'ssd_mobilenet_v1_coco': {
        'input_pb':   os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_mobilenet_v1_coco.pb')),
        'tmp_uff':    os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_mobilenet_v1_coco.uff')),
        'output_bin': os.path.abspath(os.path.join(
                          DIR_NAME, 'TRT_ssd_mobilenet_v1_coco.bin')),
        'num_classes': 91,
        'min_size': 0.2,
        'max_size': 0.95,
        'input_order': [0, 2, 1],  # order of loc_data, conf_data, priorbox_data
    },
    'ssd_mobilenet_v1_egohands': {
        'input_pb':   os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_mobilenet_v1_egohands.pb')),
        'tmp_uff':    os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_mobilenet_v1_egohands.uff')),
        'output_bin': os.path.abspath(os.path.join(
                          DIR_NAME, 'TRT_ssd_mobilenet_v1_egohands.bin')),
        'num_classes': 2,
        'min_size': 0.05,
        'max_size': 0.95,
        'input_order': [0, 2, 1],  # order of loc_data, conf_data, priorbox_data
    },
    'ssd_mobilenet_v2_coco': {
        'input_pb':   os.path.abspath(os.path.join(
                          DIR_NAME, 'frozen_inference_graph.pb')),
        'tmp_uff':    os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_mobilenet_v2_coco.uff')),
        'output_bin': os.path.abspath(os.path.join(
                          DIR_NAME, 'TRT_ssd_mobilenet_v2_coco.bin')),
        'num_classes': 4,
        'min_size': 0.2,
        'max_size': 0.95,
        'input_order': [0, 2, 1],  # order of loc_data, conf_data, priorbox_data
    },
    'ssd_mobilenet_v2_egohands': {
        'input_pb':   os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_mobilenet_v2_egohands.pb')),
        'tmp_uff':    os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_mobilenet_v2_egohands.uff')),
        'output_bin': os.path.abspath(os.path.join(
                          DIR_NAME, 'TRT_ssd_mobilenet_v2_egohands.bin')),
        'num_classes': 91,
        'min_size': 0.05,
        'max_size': 0.95,
        'input_order': [1, 0, 2],  # order of loc_data, conf_data, priorbox_data
    },
[... 205 lines cut ...]
        spec)
    _ = uff.from_tensorflow(
        dynamic_graph.as_graph_def(),
        output_nodes=['NMS'],
        output_filename=spec['tmp_uff'],
        text=True,
        debug_mode=DEBUG_UFF)
    with trt.Builder(TRT_LOGGER) as builder, builder.create_network() as network, trt.UffParser() as parser:
        builder.max_workspace_size = 1 << 28
        builder.max_batch_size = 1
        builder.fp16_mode = True

        parser.register_input('Input', INPUT_DIMS)
        parser.register_output('MarkOutput_0')
        parser.parse(spec['tmp_uff'], network)
        engine = builder.build_cuda_engine(network)

        buf = engine.serialize()
        with open(spec['output_bin'], 'wb') as f:
            f.write(buf)

if __name__ == '__main__':
    main()

```

jkjung-avt, 2022-05-21:
I think the failed assertion statement is [this line](https://github.com/NVIDIA/TensorRT/blob/release/7.0/plugin/nmsPlugin/nmsPlugin.cpp#L246):

```
ASSERT(numPriors * numLocClasses * nbBoxCoordinates == inputDims[param.inputOrder[0]].d[0]);
```

If you have made sure you have the right `num_classes`, I think you should double check that you've set `inputOrder` correctly.  Please read https://github.com/jkjung-avt/tensorrt_demos/issues/111#issuecomment-630265855 for how to find the correct order.

jkjung-avt, 2022-06-09:
Has the issue been resolved?

kanakedge, 2022-06-21:
I am also facing the same issue. 
            `'input_order': [0, 2, 1],  # order of loc_data, conf_data, priorbox_data`

jkjung-avt, 2022-06-21:
@wadhwasahil @kanakedge  Do you see a "ssd_mobilenet_v2_coco.pbtxt" file generated by "build_engine.py"?  Could you post the content of that file?

kanakedge, 2022-06-21:
Yes, "ssd_mobilenet_v2_coco.pbtxt" is generated.
`"""build_engine.py

This script converts a SSD model (pb) to UFF and subsequently builds
the TensorRT engine.

Input : ssd_mobilenet_v[1|2]_[coco|egohands].pb
Output: TRT_ssd_mobilenet_v[1|2]_[coco|egohands].bin
"""

import os
import ctypes
import argparse
import yaml

import numpy as np
import uff
import tensorrt as trt
import graphsurgeon as gs
import logging

from SimpleCalibrator import SimpleCalibrator

logging.basicConfig(level=logging.DEBUG,
                    format="%(asctime)s - %(name)s - %(levelname)s - %(message)s",
                    datefmt="%Y-%m-%d %H:%M:%S")
logger = logging.getLogger(__name__)

DIR_NAME = "ssd_mobilenet_v1_coco_2018_01_28"
NUM_CLASSES = 90
BATCHES = [1, 2, 4, 8]
# with open("/input_models/config.yaml") as f:
#     config_yaml = yaml.load(f, Loader=yaml.FullLoader)
# NUM_CLASSES = len(config_yaml["class_names"])
LIB_FILE = os.path.abspath(os.path.join(DIR_NAME, 'libflattenconcat.so'))

models = ['ssd_mobilenet_v1_coco', 'ssd_mobilenet_v2_coco', 'ssd_inception_v2_coco', 'ssdlite_mobilenet_v2_coco']

def model_specs(DIR_NAME, model, modelfile):
    MODEL_SPECS = {
        'ssd_mobilenet_v1_coco': {
            'input_pb':   os.path.abspath(os.path.join(
                            DIR_NAME, modelfile)),
            'tmp_uff':    os.path.abspath(os.path.join(
                            DIR_NAME, model+'.uff')),
            'output_bin': os.path.abspath(os.path.join(
                            DIR_NAME, model+'.enp')),
            'num_classes': 91,
            'min_size': 0.2,
            'max_size': 0.95,
            'input_order': [0, 2, 1],  # order of loc_data, conf_data, priorbox_data
        },
        'ssd_mobilenet_v2_coco': {
            'input_pb':   os.path.abspath(os.path.join(
                            DIR_NAME, modelfile)),
            'tmp_uff':    os.path.abspath(os.path.join(
                            DIR_NAME, model+'.uff')),
            'output_bin': os.path.abspath(os.path.join(
                            DIR_NAME, model+'.bin')),
            'num_classes': 91,
            'min_size': 0.2,
            'max_size': 0.95,
            'input_order': [1,0,2],  # order of loc_data, conf_data, priorbox_data
        },
        'ssd_inception_v2_coco': {
            'input_pb':   os.path.abspath(os.path.join(
                            DIR_NAME, modelfile)),
            'tmp_uff':    os.path.abspath(os.path.join(
                            DIR_NAME, model+'.uff')),
            'output_bin': os.path.abspath(os.path.join(
[... 300 lines cut ...]
                logger.info("Setting {}".format(builder_flag_map[flag]))
                config.set_flag(builder_flag_map[flag])
            
            logger.info("Building Engine...")
            parser.register_input('Input', INPUT_DIMS)
            parser.register_output('MarkOutput_0')
            parser.parse(spec['tmp_uff'], network)

            print("its builder_here!",builder)        
            engine = builder.build_engine(network,config)
            buf = engine.serialize()

            with open(os.path.join(DIR_NAME, model+f"_bs{bs}_"+f"_{quantization}_"+".enp"), 'wb') as f:
                f.write(buf)

            # if(args.quantization=="fp16"):
                # with open(os.path.join(DIR_NAME, model+f"_bs{bs}_"+f"_{quantization}_"+".enp"), 'wb') as f:
                    # f.write(buf)
            # else:
            #     with open(os.path.join(DIR_NAME, model+"_"+quantization+"_"+".enp"), 'wb') as f:
            #         f.write(buf)

if __name__ == '__main__':
    main()
`

kanakedge, 2022-06-21:
> @wadhwasahil @kanakedge Do you see a "ssd_mobilenet_v2_coco.pbtxt" file generated by "build_engine.py"? Could you post the content of that file?

Yes "ssd_mobilenet_v2_coco.pbtxt" is generated

kanakedge, 2022-06-22:
> @kanakedge Please share the content of "ssd_mobilenet_v2_coco.pbtxt"? It would help to debug the issue.

Please check this drive link: 
https://drive.google.com/drive/folders/1GxRNlUU_KwIFk-pLxEBd_iSK-5H2h2QM?usp=sharing

kanakedge, 2022-06-22:
> > I am also facing the same issue. `'input_order': [0, 2, 1], # order of loc_data, conf_data, priorbox_data`
> 
> This actually worked for me.

Can you share your build._engine.py?

jkjung-avt, 2022-07-03:
@kanakedge  Sorry for my late reply.

> Please check this drive link:
> https://drive.google.com/drive/folders/1GxRNlUU_KwIFk-pLxEBd_iSK-5H2h2QM?usp=sharing

The key lines to note in this "ssd_mobilenet_v2_coco.pbtxt" are the `inputs` to the `NMS` node:

```
  nodes {
    id: "NMS"
    inputs: "concat_box_conf"
    inputs: "Squeeze"
    inputs: "concat_priorbox"
    operation: "_NMS_TRT"
```

Also referring to "inputOrder" in [documentation of the NMS plugin](https://github.com/NVIDIA/TensorRT/tree/main/plugin/nmsPlugin#parameters):

> Specifies the order of inputs {loc_data, conf_data, priorbox_data}, in other words, inputOrder[0] is for loc_data, inputOrder[1] is for conf_data and inputOrder[2] is for priorbox_data. For example, if your inputs in the memory are in the order of loc_data, priorbox_data, conf_data, then inputOrder should be [0, 2, 1].  

So the correct setting of "inputOrder" for your "ssd_mobilenet_v2_coco" model should be: [1, 0, 2].

jkjung-avt, 2022-07-17:
Close due to no response.

closed 2022-07-17 by jkjung-avt