jkjung-avt/tensorrt_demos #554: nvcc fatal   : Unsupported gpu architecture 'compute_N'

Md7-jx, 2022-05-17:
Hi, Jkjung. 
Thanks for this project ,I succeeded it in ubuntu16.04 , but in Windows10 I failed. When I run "make" command under the folder "plugins/" ,I got **computes: N o n e.**
 "computes: N o n e
NVCCFLAGS: -gencode arch=compute_N,code=[sm_N,compute_N] -gencode arch=compute_o,code=[sm_o,compute_o] -gencode arch=compute_n,code=[sm_n,compute_n] -gencode arch=compute_e,code=[sm_e,compute_e]
nvcc -ccbin g++ -I"C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v11.1/include" -I"G:/test/TensorRT-8.2.4.2/include" -I"/usr/local/include" -I"plugin" -gencode arch=compute_N,code=[sm_N,compute_N] -gencode arch=compute_o,code=[sm_o,compute_o] -gencode arch=compute_n,code=[sm_n,compute_n] -gencode arch=compute_e,code=[sm_e,compute_e] -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
nvcc fatal   : Unsupported gpu architecture 'compute_N'
Makefile:43: recipe for target 'yolo_layer.o' failed
mingw32-make: *** [yolo_layer.o] Error 1 "
My gpu is  RTX 3060. How to solve this problem?

jkjung-avt, 2022-05-17:
Which GPU are you using?

The problem is that [gpu_cc.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/plugins/gpu_cc.py) fails to identify the model of GPU...

Md7-jx, 2022-05-18:
> Which GPU are you using?
> 
> The problem is that [gpu_cc.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/plugins/gpu_cc.py) fails to identify the model of GPU...

The GPU is GeForce RTX 3090.
I get something wrong with
 `libnames = ('libcuda.so', 'libcuda.dylib', 'cuda.dll')    for libname in libnames:
        try:
            cuda = ctypes.CDLL(libname)
        except OSError:
            continue
        else:
            break
    else:
        return`
There comes **OSError**，maybe I lost these 3 files.

jkjung-avt, 2022-05-18:
My "libcuda.so" is located at `/usr/lib/x86_64-linux-gnu/libcuda.so`.  I think it was installed when I installed NVIDIA driver on the system.

Md7-jx, 2022-05-18:
Thanks for anwsering, I will try to update my NVIDIA driver .

closed 2022-05-21 by jkjung-avt