jkjung-avt/tensorrt_demos #557: custom model to trt-int8 problem

tingggggg, 2022-05-27:
Hi,
I converted own yolo model (onnx) to mymodel.trt(FP16) & mymodel-int8.trt(INT8) then use trt_yolo.py to observe the detection results.
With same image, FP16 model could detect correctly objects but not INT8 model.

My model is a mobilenet arch yolo (two ouputs).
I calibrated model with about 2,700 images.

Did I missing any detail to lead this error result?

Thank you~

jkjung-avt, 2022-05-31:
I have no idea about why.  I also observed big mAP discrepency between FP16 and INT8 TensorRT engines for the "yolov4-608" model.

You could probably try to use polygraphy to compare inference results: ONNX vs. TensorRT FP16 vs. TensorRT INT8.  Reference: https://github.com/jkjung-avt/tensorrt_demos/issues/315#issuecomment-817143067

jkjung-avt, 2022-06-16:
@tingggggg  Does your mobilenet-backboned yolo model require somewhat different preprocessing than what's implemented in this repo?  Maybe it's a good idea for you to double check the preprocessing function as well?

Please refer to: https://github.com/jkjung-avt/tensorrt_demos/issues/565#issuecomment-1157632363

closed 2022-06-27 by jkjung-avt