jkjung-avt/tensorrt_demos #558: (QUESTION) Which model to use for multipas detections

MatheusCaltran, 2022-06-02:
Hello JK Jung

With your repositories I can learn more about AI, so I have a lot of consideration for you

I would like to know in your opinion which model I can use to detect several objects?(as shown in the image I have to do the cane detection in an agricultural machine)

[image]
[image]
[image]

MatheusCaltran, 2022-06-02:
thank you I hope one day to have a little of your knowledge, anything I repost the topic to ask about something else. But I would try to avoid it so as not to occupy your time.

agjunyent, 2022-06-03:
If you only need to detect the canes (1 class), in this environment, yolov4-tiny-3l from darknet (https://github.com/AlexeyAB/darknet) will be more than sufficient, with good fps. If you increase the number of classes but maintain the "aspect ratio" of them (so all classes have similar sized boxes), yolov4-tiny-3l still is a very good candidate.

Crucial parts:
- Labelling, as @jkjung-avt said. I'd say that 50% of the success of a model, or even more, is training dataset labels quality
- Number of training samples. The more the better, but slower to label
- mAP requierements. You need to think that you will never have 100% accuracy / mAP.

MatheusCaltran, 2022-06-05:
thank you very much, i will try with yolov4-tiny-3l sorry to take up your time

MatheusCaltran, 2022-07-15:
thanks I will focus on using CSI, I will use ECAM-30 for now

MatheusCaltran, 2022-12-02:
Hi @kjung-avt how are you?

we have already made several modifications to the machine, I am currently using jetson nano, e-cam30, yolov4-tiny

detection photo:
[image]
[image]
[image]
[image]

but now we do a test simulating reality and it wasn't very lucky because there was straw, but this will only happen in extreme cases because of the rain that makes the straw stick to the cane

example:
[image]

[image]
[image]
[image]

With all this I just want to know about your opinion what can I do to improve this? for me your opinion is very important, much of what I can do here was because of you.

jkjung-avt, 2022-12-03:
Adding pictures with straws into your training set would definitely help.  I think you should do that.

I think a trained model should be able to detect sugarcanes which are partially covered by straws and still clearly visible.

closed 2023-04-06 by jkjung-avt