jkjung-avt/tensorrt_demos #562: WARNING:root:Camera: starting while cap is not opened! ERROR: failed to open camera!

Ardihermanto, 2022-06-10:
Hello..
I want to ask about TensorRT Demos, when I ran the program with the RTSP link.. the output error was like the below..

**WARNING:root:Camera: starting while cap is not opened!
ERROR: failed to open camera!**

Why the program error?

Thank you

jkjung-avt, 2022-06-11:
Please refer to my blog post: [How to Capture and Display Camera Video with Python on Jetson TX2](https://jkjung-avt.github.io/tx2-camera-with-python/)

Make sure you've installed all necessary GStreamer components for RTSP.

ilhamrayhan117, 2022-06-11:
hello, i've run into the same problem while using the usb argument on trt_yolo, this is the command i'm using
python3 trt_yolo.py --usb 0 -m yolov4_tsr_v1

and this is the output 
WARNING:root:Camera: starting while cap is not opened!
ERROR: failed to open camera

please help, thank you

jkjung-avt, 2022-06-12:
@ilhamrayhan117  Try setting `USB_GSTREAMER` to `False` to see if it works for you. 

https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/utils/camera.py#L22

Ardihermanto, 2022-06-12:
> Please refer to my blog post: [How to Capture and Display Camera Video with Python on Jetson TX2](https://jkjung-avt.github.io/tx2-camera-with-python/)
> 
> Make sure you've installed all necessary GStreamer components for RTSP.

hello Jiun,

I have not found a solution for gstreamer rtsp

jkjung-avt, 2022-06-12:
@Ardihermanto  Please refer to https://github.com/jkjung-avt/tensorrt_demos/issues/220 to debug the issue.  Basically you need to make sure you could access your RTSP stream by GStreamer first.

Ardihermanto, 2022-06-12:
Thank you Jiun for your information

closed 2022-06-27 by jkjung-avt