jkjung-avt/tensorrt_demos #566: Convert SSD model to ONNX instead of UFF

wadhwasahil, 2022-06-17:
I am using your code to convert my trained SSD MobileNet V2 model to Tensor RT format. You have used UFF format - https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/ssd/build_engine.py#L282

However I want to use Onnx instead of UFF. Once the graph is generated how can I convert it to ONNX instead of UFF?

Thanks

jkjung-avt, 2022-06-17:
Please refer to the official sample by NVIDIA: https://github.com/NVIDIA/TensorRT/tree/main/samples/python/tensorflow_object_detection_api

wadhwasahil, 2022-06-17:
So the thing is I have used  `tf.train.write_graph(dynamic_graph.as_graph_def(), ".","test.pb", as_text=False)` to get a .pb file. However I am unable to load the .pb file. Is this a correct way?

jkjung-avt, 2022-06-17:
I recommend to use [export_inference_graph.py](https://github.com/tensorflow/models/blob/master/research/object_detection/export_inference_graph.py) to export your custom trained model to pb.  The resulting pb file would be named as "frozen_inference_graph.pb".

wadhwasahil, 2022-06-17:
And then you're saying to use your code to change plugins etc?

closed 2022-06-27 by jkjung-avt