jkjung-avt/tensorrt_demos #567: eval_ssd set up question

hotudon, 2022-06-18:
Hi,

I am trying to test the eval ssd demo to run following your tutorial.  I was unable to use sudo pip3 install pycocotools to install pycocotools, so i decide to clone the COCOAPI and build it from the source.  It didn't look like it failed using that route, but I am running into this error when I try to run the python3 eval_ssd.py command

Traceback (most recent call last):
  File "eval_ssd.py", line 16, in <module>
    from pycocotools.coco import COCO
ModuleNotFoundError: No module named 'pycocotools

Did i fail to install pycocotool?

Thanks.

jkjung-avt, 2022-06-19:
I guess so.  You need to install "pycocotools".  Please refer to: https://github.com/jkjung-avt/tensorrt_demos/blob/master/README_mAP.md

hotudon, 2022-06-19:
I manage to get the pycocotools to install this time around using git repo.  I saw an previous issue with making changes to /cocoeval.py#L506

SSD eval ran this time.  Thanks again for putting the tutorials up.  I will try the YOLO eval next

closed 2022-06-21 by hotudon