jkjung-avt/tensorrt_demos #574: WARNING:root:Camera: starting while cap is not opened!

starlord263, 2022-07-22:
Hi, Good evening i am facing following error while passing --rtsp stream:

WARNING:root:Camera: starting while cap is not opened!

ERROR: failed to open camera!

I've tried all available solutions in github issues but still unable to resolve.

i am passing  python3 trt_mtcnn.py --rtsp rtsp://root:esec@192.168.1.204/axis-media/media.amp/
rtsp://root:admin@192.168.1.200/axis-media/media.amp/

I also tried using this command gst-launch-1.0 -e rtspsrc location="rtsp://root:admin@192.168.1.200/axis-media/media.amp" ! rtph264depay ! h264parse ! omxh264dec ! nveglglessink and it is working fine i am able to see output.

But i don't know why can not pass it as argument.

jkjung-avt, 2022-07-22:
As a quick check, could you try to see if setting `--rtsp_latency 2000` (command-line option) fixes the problem for you?

jkjung-avt, 2022-08-19:
Closing due to no further response.

closed 2022-08-19 by jkjung-avt