jkjung-avt/tensorrt_demos #576: failed to build the TensorRT engine

soukaina11111, 2022-07-26:
Hi @jkjung-avt  first, thank you for your amazing work,  

I have tested this demo on jetson tx2 and it’s work! and i wanted to test it on jetson xavier, i succefully convert yolov4.weights to yolov4.onnx, but when I try to convert onnx to tensorrt without int8 and dla_core i got this error :

`python onnx_to_tensorrt.py -m yolov4
Loading the ONNX file...
Adding yolo_layer plugins.
Adding a concatenated output as "detections".
Naming the input tensort as "input".
Building the TensorRT engine.  This would take a while...
(Use "--verbose" or "-v" to enable verbose logging.)
onnx_to_tensorrt.py:148: DeprecationWarning: Use set_memory_pool_limit instead.
  config.max_workspace_size = 1 << 30
onnx_to_tensorrt.py:170: DeprecationWarning: Use build_serialized_network instead.
  engine = builder.build_engine(network, config)
[07/26/2022-11:14:38] [TRT] [W] Convolution + generic activation fusion is disable due to incompatible driver or nvrtc
[07/26/2022-11:14:39] [TRT] [E] 1: [ltWrapper.cpp::CublasLtWrapper::66] Error Code 1: Cublas (Could not initialize cublasLt. Please check CUDA installation.)
ERROR: failed to build the TensorRT engine!`

Any idea how to solve it?
best regards.

jkjung-avt, 2022-07-27:
This might be a problem of the specific combination CUDA 10.2 and TensorRT 7.2.1.x?  Please refer to: https://github.com/NVIDIA/TensorRT/issues/866

Please test with the latest version of JetPack.

soukaina11111, 2022-07-28:
Thank you for your answer .I’m using jetpack 5.0.1, cuda 11.4  and tensorrt 8.4.0.11

jkjung-avt, 2022-07-28:
Then I have no idea.  The error message points to a problem about the [cuBLAS](https://docs.nvidia.com/cuda/cublas/index.html) library.  Please direct your question to NVIDIA instead.

> [07/26/2022-11:14:39] [TRT] [E] 1: [ltWrapper.cpp::CublasLtWrapper::66] Error Code 1: Cublas (Could not initialize cublasLt.
> 
> Please check CUDA installation.)

closed 2022-08-19 by jkjung-avt

soukaina11111, 2022-08-25:
Problem solved by reflashing the os, and installing the newest version of jetpack 5.0.2.
 thanks again :).

jkjung-avt, 2022-08-25:
Good to hear that.