jkjung-avt/tensorrt_demos #577: Run trt_yolo on batch of image

remiruzn, 2022-08-17:
How i can run **trt_yolo.detect(frame, conf_th)** on multiple image like a batch. 
Need something like 
```
frames=[frame_1, frame_2, frame_3]
trt_yolo.detect(frames, conf_th)
```
I have problem with optimal inference trt_yolo on 3 cameras in real time, can some one help me, any advice appreciated.

In general i have something like this:
```

while True:
        ret1, frame1 = cap1.read()
        ret2, frame2 = cap2.read()
        ret3, frame3 = cap3.read()
        if frame1 is None:  break
        if frame2 is None:  break
        if frame3 is None:  break

        boxes1, confs1, clss1 = trt_yolo.detect(frame1, conf_th)
        boxes2, confs2, clss2 = trt_yolo.detect(frame2, conf_th)
        boxes2, confs2, clss2 = trt_yolo.detect(frame3, conf_th)
        
        # Some logic with boxes
```

jkjung-avt, 2022-08-19:
Please refer to: https://github.com/jkjung-avt/tensorrt_demos/issues/306#issuecomment-940579878.

remiruzn, 2022-08-19:
Is this best practice for work with several cameras?

jkjung-avt, 2022-08-19:
> Is this best practice for work with several cameras?

That depends on your application.  But you might want to look into [NVIDIA DeepStream SDK](https://developer.nvidia.com/deepstream-sdk).

remiruzn, 2022-08-19:
Ok. Thanks for help!

closed 2022-08-19 by remiruzn