jkjung-avt/tensorrt_demos #578: Issue building yolo plugin with TRT 6.5

lazatsoc, 2022-08-24:
Hi, 

Thanks for your useful contribution.

I am trying to run Yolo (Demo 4,5) on x86_64 machine without success. I use a GTX 1080Ti (CC=61), TensorRT version 6.5 and CUDA 10.2. According to prerequisites, Yolo should run with this configuration.

When trying to build the yolo-layer plugin, I get kind of the same error with 
https://github.com/jkjung-avt/tensorrt_demos/issues/576, https://github.com/jkjung-avt/tensorrt_demos/issues/222#issuecomment-732463357. The output of `make` is shown here: https://pastebin.com/QeJnYUPh.

I have given the correct CC in Makefile:

`compute=61`

I have ensured that the libraries and the headers of TRT can be found, changing the paths in the Makefile like this:

`TENSORRT_INCS=-I"/usr/include/x86_64_linux_gnu"`
`TENSORRT_LIBS=-I"/usr/lib/x86_64_linux_gnu`.

The errors seem like an incompatibility with the headers of TensorRT 6.5. 

Is it true that Yolo requires 6.x+ and not 7.x+?
Is there a workaround for these errors?

jkjung-avt, 2022-08-24:
Errors like this one could be fixed easily by modifying function declarations in the header and C++ source files.

> yolo_layer.h(91): error: exception specification for virtual function "nvinfer1::YoloLayerPlugin::detachFromContext" is incompatible with that of overridden function "nvinfer1::IPluginV2Ext::detachFromContext"

I'm not able to find TensorRT 6.5 header files from NVIDIA web site (https://developer.nvidia.com/nvidia-tensorrt-6x-download) so I don't know what exactly went wrong for you.

Just as a quick check, could you remove the following lines of code.

   https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/plugins/yolo_layer.h#L13-L17

And replace them with 1 single line of:  (force `NOEXCEPT` to be empty)

   `#define NOEXCEPT`

Then try to make the plugin again.

lazatsoc, 2022-08-25:
Thanks for your quick reply.

I managed to build `libyolo_layer.so`. The solution was to replace all conditions with `NV_TENSORRT_MAJOR` and to keep the configuration for TRT 8. I don't know why this is correct for TRT 6.5 that I use. This version is found on DRIVE AGX and nowhere else, so maybe it is a special case.

Specifically, I did the following changes:
1) I replaced https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/plugins/yolo_layer.h#L13-L17 with https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/plugins/yolo_layer.h#L14
2) I replaced  https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/plugins/yolo_layer.h#L93-L97 with https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/plugins/yolo_layer.h#L94
3) I replaced https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/plugins/yolo_layer.cu#L260-L264 with https://github.com/jkjung-avt/tensorrt_demos/blob/a061e44a82e1ca097f57e5a32f20daf5bebe7ade/plugins/yolo_layer.cu#L261

jkjung-avt, 2022-08-25:
OK.  Thanks for sharing.

closed 2022-08-26 by lazatsoc