jkjung-avt/tensorrt_demos #587: Error when install install_pycuda.sh file

gurselturkeri, 2023-01-15:
Hi, I would like to install `install_pycuda.sh` but I get error like this:
`/usr/bin/ld: cannot find -lboost_python*-py3?
collect2: error: ld returned 1 exit status
error: command 'x86_64-linux-gnu-g++' failed with exit status 1
make: *** [Makefile:6: all] Error 1
`
**Python: 3.8.10
Cuda: 11.6
TensorRT:  8.4.2-1**

bredebjorhovd, 2023-01-25:
I had the same issue. If you look at line 21 this refers to the boost_pylib path, so you need to find where this is on your computer.
To exemplify, heres how I changed my path

`boost_pylib=$(basename /usr/lib/${arch}-linux-gnu/libboost_python38.so)`

Hope this helps.

jkjung-avt, 2023-04-06:
Same issue as https://github.com/jkjung-avt/tensorrt_demos/issues/595.

Hopefully it has been fixed by https://github.com/jkjung-avt/tensorrt_demos/commit/107b36af74f6f89c9e392c9cc54da149846bda3f.

closed 2023-04-06 by jkjung-avt