jkjung-avt/tensorrt_demos #592: Failure to Build ONNX from Custom Yolo

DJT777, 2023-02-07:
Hi,

I'm running into this issue when trying to build my onnx engine from a custom yolo cofig+weights.

```
(deepsort_env) user@nano:~/tensorrt_demos/yolo$ python yolo_to_onnx.py -m MK2ieee -c 2
Parsing DarkNet cfg file...
Building ONNX graph...
Traceback (most recent call last):
  File "yolo_to_onnx.py", line 1014, in <module>
    main()
  File "yolo_to_onnx.py", line 1002, in main
    verbose=True)
  File "yolo_to_onnx.py", line 516, in build_onnx_graph
    major_node_specs = self._make_onnx_node(layer_name, layer_dict)
  File "yolo_to_onnx.py", line 589, in _make_onnx_node
    node_creators[layer_type](layer_name, layer_dict)
  File "yolo_to_onnx.py", line 836, in _make_route_node
    assert route_node_specs.channels % groups == 0
AssertionError
```
Using this config
```
[net]
# Testing
#batch=1
#subdivisions=1
# Training
batch=64
subdivisions=16
width=512
height=512
channels=3
momentum=0.9
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.00261
burn_in=1000
max_batches = 500200
policy=steps
steps=400000,450000
scales=.1,.1

[convolutional]
batch_normalize=1
filters=21
size=3
stride=2
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=21
size=3
stride=2
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[route]
layers=-1
groups=2
group_id=1

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[route]
layers = -1,-2

[convolutional]
batch_normalize=1
filters=21
size=1
stride=1
pad=1
activation=leaky

[route]
layers = -6,-1

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[route]
layers=-1
groups=2
group_id=1

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[route]
layers = -1,-2

[convolutional]
batch_normalize=1
filters=21
size=1
stride=1
pad=1
activation=leaky

[route]
layers = -6,-1

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[route]
layers=-1
groups=2
group_id=1

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[route]
layers = -1,-2

[convolutional]
batch_normalize=1
filters=21
size=1
stride=1
pad=1
activation=leaky

[route]
layers = -6,-1

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

##################################

[convolutional]
batch_normalize=1
filters=21
size=1
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[convolutional]
size=1
stride=1
pad=1
filters=21
activation=linear

[yolo]
mask = 3,4,5
anchors = 10,14,  23,27,  37,58,  81,82,  135,169,  344,319
classes=2
num=6
jitter=.3
scale_x_y = 1.05
cls_normalizer=1.0
iou_normalizer=0.07
iou_loss=ciou
ignore_thresh = .7
truth_thresh = 1
random=0
resize=1.5
nms_kind=greedynms
beta_nms=0.6

[route]
layers = -4

[convolutional]
batch_normalize=1
filters=21
size=1
stride=1
pad=1
activation=leaky

[upsample]
stride=2

[route]
layers = -1, 23

[convolutional]
batch_normalize=1
filters=21
size=3
stride=1
pad=1
activation=leaky

[convolutional]
size=1
stride=1
pad=1
filters=21
activation=linear

[yolo]
mask = 0,1,2
anchors = 10,14,  23,27,  37,58,  81,82,  135,169,  344,319
classes=2
num=6
jitter=.3
scale_x_y = 1.05
cls_normalizer=1.0
iou_normalizer=0.07
iou_loss=ciou
ignore_thresh = .7
truth_thresh = 1
random=0
resize=1.5
nms_kind=greedynms
beta_nms=0.6
```

jkjung-avt, 2023-02-08:
```
    assert route_node_specs.channels % groups == 0
```

Here's the reason of this assertion error.

* The "route" layers in the cfg try to split the tensor into 2 groups.

   ```
   [route]
   layers=-1
   groups=2
   group_id=1
   ```

* However, the previous conv layer outputs are with 21 filters (or channels).

   ```
   filters=21
   ```

* 21 is not a multiple of 2 (21 cannot be split into 2 groups equally).  My code does not know how to handle such cases and throws the assertion error.

DJT777, 2023-02-08:
@jkjung-avt 

Thanks for clarifying that.

closed 2023-02-08 by DJT777