jkjung-avt/tensorrt_demos #594: wrong many bounding box  and wrong predict untrained class id

MehmetOKUYAR, 2023-02-09:
[image]

It's predicting the class id that I haven't trained, and it's predicting a lot of random bounding grays. I didn't know where there was an error. It works fine when I try it on another pc.

xiaoyaoguai, 2024-06-28:
> [image]
> 
> It's predicting the class id that I haven't trained, and it's predicting a lot of random bounding grays. I didn't know where there was an error. It works fine when I try it on another pc.
> 
> weights work fine if i don't do tensorrt transform with opencv darknet library

You can try modifying the py file of class_id.
I encountered a similar issue on Nano, where TensorRT's inference was entirely incorrect, and the generated Tiny model was 32mb (compared to the correct model generated on other devices, which is only 13mb). Notably, Jetpack 4.6 comes pre-configured with Python 3.6 and Protoc 3.0.0, and upgrading Protoc from 3.0.0 to 3.8 renders ONNX 1.9 unimportable. I attempted upgrading Protoc after importing ONNX, but this still did not resolve the issue.

xiaoyaoguai, 2024-07-01:
try this：rebuild the "yolo_layer.h" in plugin,      
line70:  bool supportsFormatCombination(int pos, const PluginTensorDesc* inOut, int nbInputs, int nbOutputs) const NOEXCEPT override { return inOut[pos].format == PluginFormat::kLINEAR && inOut[pos].type == DataType::kFLOAT; }