jkjung-avt/tensorrt_demos #595: error happening when sh ./install_pycuda.sh

ZnSn399379, 2023-03-29:
WARNING: The directory '/home/jetson/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag.
Requirement already satisfied: setuptools in /usr/local/lib/python3.6/dist-packages (57.4.0)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
./install_pycuda.sh: 23: ./install_pycuda.sh: Bad substitution

ZnSn399379, 2023-03-30:
It output like this
ls: cannot access '/usr/lib/-linux-gnu/libboost_python*': No such file or directory

jkjung-avt, 2023-04-06:
I just committed a fix to install-pycuda.sh.  Please give it a try.  Thanks.