jkjung-avt/tensorrt_demos #598: Trouble with linker on yolo plugin x86

sgriset, 2023-04-08:
I'm building the yolo plugin for x86 compute architecture 86 (Cuda 11.8 & TensorRT-8.5.2.2).  The compile has a if few warning but the linker is giving me the following message:

/usr/bin/ld: skipping incompatible /usr/local/TensorRT-8.5.2.2/lib/libnvinfer.so when searching for -lnvinfer
/usr/bin/ld: skipping incompatible /usr/local/TensorRT-8.5.2.2/lib/libnvparsers.so when searching for -lnvparsers
/usr/bin/ld: skipping incompatible /usr/local/TensorRT-8.5.2.2/lib/libnvinfer_plugin.so when searching for -lnvinfer_plugin

Any thoughts?

jkjung-avt, 2023-04-08:
Could this be a problem of mismatching TensorRT and CUDA libraries?  If that's the case, you should remove the incompatible library and install a proper one instead.

Take "TensorRT 8.5 GA Update 2" as an example, NVIDIA offers the libraries for either "CUDA 11.x" or "CUDA 10.2".  Make sure you choose the correct one.

[image]

sgriset, 2023-04-13:
You were correct there was some mix up in libraries (cuda and tensorrt).   Since the server I'm using is used by several developers I decided to use a cuda and tensorrt container from NVIDIA's catalog.  The plugin compiled and linked cleanly.  Thank you for your help.

closed 2023-05-02 by sgriset