jkjung-avt/tensorrt_demos #601: How to prevent tensorrt from fusing the final layers when trying to convert tiny yolov3 to int8?

NIKHILDUGAR, 2023-04-25:
I used the example given in https://github.com/jkjung-avt/tensorrt_demos#demo-5-yolov4 to convert my tiny yolov3 to int8 but the issue is that when I use --verbose on onnx_to_tensorrt.py step I see that the final layers are being fused.
How do I stop this from happening?

Thanks

jkjung-avt, 2023-04-26:
In general, you could do [mark_output()](https://docs.nvidia.com/deeplearning/tensorrt/api/python_api/infer/Graph/Network.html#tensorrt.INetworkDefinition.mark_output) on a specific layer to prevent TensorRT from fusing it with other layers.  Reference: https://github.com/NVIDIA/TensorRT/issues/252#issuecomment-577468499

Otherwise, please be more specific about which layers in your tiny yolov3 model you're having the trouble.

NIKHILDUGAR, 2023-04-26:
Hey, thanks for the quick response. I did look at the reference you provided and the problem I am having with it is that it doesn't really take layer names only tensors and the method I have seen mostly used by people is:- 

`       
last_layer = network.get_layer(network.num_layers -1)
network.mark_output(last_layer.get_output(0))
`
Now since the output layer of tiny yolo ( in my case) or yolo is on different levels I can't correctly identify which layers to mark and how to do so correctly.